scrapcomputing/ThrottleBlaster #1: LEDC68 Display Compatibility

skadarnold, 2024-04-07:
Thank you for your work on this, I am very excited to give it a try. 

I don't see myself having as much use for a 4 digit display, and I have a couple of LEDC68's around from some gotek floppy drives. 

https://github.com/coopzone-dc/GotekLEDC68

It is based on the TM1651 so I wonder if it would be compatible without any modification. I plan to give a try when I have everything built, but I thought I might bring it to your attention.

scrapcomputing, 2024-04-07:
This is the exact display I used in the first video :) It works, but not as well as the 4-digit TM1637: (i)  the display won't show the rightmost digit, and (ii) it shows some erroneous dots.

To fix issue (i) you can build the firmware  using the additional configuration: `-DDISPLAY_SHIFT_LEFT=1` in `cmake -DDISPLAY_SHIFT_LEFT=1 -DCMAKE_BUILD_TYPE=Release -DPICO_SDK_PATH=/path/to/pico-sdk/ ../src/`. Please try the attached firmware, which is based on rev.0.3 and is configured with this option.

I am not sure how to fix issue (ii), as I am relying on a third-party library for driving the display, and I couldn't figure out how to fix it.
[ThrottleBlaster_rev0.3_3digit_TM1651.uf2.zip](https://github.com/scrapcomputing/ThrottleBlaster/files/14898361/ThrottleBlaster_rev0.3_3digit_TM1651.uf2.zip)

skadarnold, 2024-04-07:
Aha I see in your video now. Thank you for the information, I might play around with some of the code and see if I can work some of that out. The 3 digit display could work particularly well for a certain case that I have if I can find a good way to mount it. 

PCB's are on their way now, but had to reorder with 0.3 update because I want to use UART. Really cool!

scrapcomputing, 2024-04-07:
Feel free to send over any patches if you manage to fix the issues.

Yeah, mounting the displays is a bit tricky, we may need a custom PCB for some of the common ones.

skadarnold, 2024-04-28:
For some reason I am stuck on building this on Windows 11. The cmake command successfully generates the configuration and it looks good, but when I run make I get "make:  *** No targets specified and no makefile found.  Stop.". I will keep playing around.

scrapcomputing, 2024-06-11:
This error message means that `Makefile` is not found in the current directory.
Are you still having this issue, or should I close the ticket?

skadarnold, 2024-06-11:
I was not able to get this to work unfortunately, but it is probably something on my end and with windows. I will have to give this a try with linux. 

Feel free to close the ticket and I will open a new issue when I have some time to work on this a bit more. Thanks for checking!

closed 2024-06-13 by scrapcomputing