scrapcomputing/ThrottleBlaster #5: MAX3232 Footprint

skadarnold, 2024-04-24:
My PCB's have finally arrived, but it seems that the IC I chose for the MAX3232 seems to be too wide. I should have looked at the specs a little closer, but I also had some confusion as to what I should be looking for. Example in the PCB layout with kicad it says it is an SOIC-16W, but it states the width at 5.3mm. Not only is SOIC-16W way too wide to even consider, but even the closest thing I can find is a SOIC-16 narrow which is 4mm wide. Most of the SOIC-16 options are around 7mm which is what I originally tried hoping the pads would line up just enough. 

Also the SOIC-16 narrow ones that may work are $7 a piece in comparison to $2. Not sure if it is really worth it at that point or not.

scrapcomputing, 2024-04-25:
That's a bummer!

The one I am using is a Ti MAX3232EC which I got locally for about $1.50. This package type is actually pretty common. If you check on Digikey these parts are listed as "16-SOIC 3.90mm width" in the Package/Case column, and the most affordable one is at $1.49 (model: MAX3232ECDR). 
Digikey's measurement refers to the IC's package width, which is indeed 3.90mm. Kicad's measurement refers to the gap between the pads on the PCB, which is 5.30mm. 

Thanks for letting me know about this, I will add a note in the BOM explaining this a bit further.
I hope you find the right part at an affordable price! Oh, and stay away of the really cheap unbranded ones, as most of them are fake chips that either don't work right away, or stop working soon after.

skadarnold, 2024-04-25:
Ok great thank you! The MAX3232EC is one that I was looking at as well since it is listed at mouser as 3.91mm width. This helps confirm this will work even though the listing specs it as SOIC-16 narrow. I had noticed in your video that the IC you used seemed slightly narrow so I thought I would take the chance with the a regular SOIC-16. I will get the new ones order and thankfully I can solder them in at a later date and build the rest for current testing.

closed 2024-04-26 by scrapcomputing