scrapcomputing/ThrottleBlaster #6: Output Noise

skadarnold, 2024-04-25:
Hello,
I was able to build my first 0.3 throttleblaster last night and everything was pretty straight forward. This morning I decided to test the output with the oscilloscope to make sure it was behaving properly before connecting it to CPU socket. The PWM seems to be behaving properly, but the line seems very noisy and unstable. There is a very strong peak on the waveform as well that makes be think it could use some filtering in general. This seems incorrect but I am not sure what is going on yet. I have checked for solder bridges and R1 and R2 are measuring in spec. I will keep you in touch but let me know if you have any suggestions. 

This happens in all modes, 1but, rot, and pot as well so I do not think it is an input / control issue. 

https://github.com/scrapcomputing/ThrottleBlaster/assets/80274795/b9738cd5-4f8d-4a0b-8ba2-52d2adc0d4e0

skadarnold, 2024-04-25:
The signal looks clean up until the MOSFET so I am going to look at that and maybe swap it out.

skadarnold, 2024-04-25:
I have swapped the MOSFET out and also tried pulling the R2 just to make sure it wasn't causing some weirdness but the problem still persists. I am thinking that things are not happy with that type of 2N7000 I used.

scrapcomputing, 2024-04-26:
You can't really measure the signal at the output pin (Q1) because the MOSFET is only pulling down the STPCLK pin, which has an internal pull up. So measuring noise at the output is expected behavior.
If you really want to measure something at the output try adding a 1K resistor to 3.3 or 5V, otherwise just measure at the MOSFET's gate.

skadarnold, 2024-04-26:
Ahh ok I see now, I was starting to wonder if this behavior was normal. Thank you for clarifying. I will give it a go with a CPU then!

closed 2024-04-26 by skadarnold