scrapcomputing/ThrottleBlaster #7: VGA port ripple current

skadarnold, 2024-04-26:
Hello again,
I have good news of success, my throttle blaster is functioning properly with a socket 7 233MMX! Very exciting to see the computer slow down to a stop with such fine control. I did want to mention that there is a slight ripple that becomes visible on my LED display as PWM is active. My graphics card is onboard motherboard so it could be situational. It is slight, but it creates a sort of wavey distortion especially across text. I will  play around with it when I have some time tomorrow.

skadarnold, 2024-04-26:
I think a lot of this is being caused by a lower quality PSU being used as my test computer. This visual distortion does not happen on my CRT.  Not sure if this is something that can be helped in the design of throttle blaster or not.

scrapcomputing, 2024-04-27:
This is quite strange. Is this distortion visible at all Throttle Blaster frequencies?

skadarnold, 2024-04-27:
Yes it is visible whenever PWM is modulating. Example if the potentiometer is all the way open it is clear, if it is all the way closed it goes away, everything else in between modulates as distortion and seems to change slightly through frequency changes. I will try to provide a video when I have a moment.

skadarnold, 2024-04-27:
Unfortunately my attempts to capture the distortion didn't work out all that well. It is easy to see with the eyes but subtle in a way.   I can also hear an oscillation happening inside of the PSU, so it could also be poor filtration there as well. It reminds me of like a coil whine from a bad fan or something like that.

scrapcomputing, 2024-04-27:
Makes sense, perhaps you could try measuring the 5V rail voltage, if that shows some ripple, then perhaps this is indeed a PSU issue.

skadarnold, 2024-04-27:
Yes you are correct. I am able to measure the ripple on the 5v line. It is probably a good idea that I replace this cheap PSU.

skadarnold, 2024-04-27:
I just tried a VGA cable with a ferrite bead and it cleared up the visual distortions. It may just be a better cable and not the bead but it definitely seems to have fixed the issue. 

The PSU whine and dirty 5v is still present aside from that though.

skadarnold, 2024-04-27:
This can be closed, but the issue is good for awareness.

closed 2024-04-27 by skadarnold