scrapcomputing/ThrottleBlaster #8: DB9 Concept Design

skadarnold, 2024-04-26:
I have been spending some time thinking of good mounting options, and I had some ideas I decided to crudely draw out that would utilize simple DB9 PCI mounting brackets like the ones shown here.. 
https://www.amazon.com/gp/product/B008635ANK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
The idea is the top ribbon DB9 would be removed then the throttle blaster could be designed to have a DB9 for direct mounting to the bracket. The DB9 could also be utilized as a pinout to send 5v, display, rotary / button combos with a simple PCB designed just for these few components. The rotary / button controls could also be combined into a ribbon cable to attach to PCB's designed for internal control like a front panel as well. Then instead of having 3 pins for RS232 connection, maybe just provide a ribbon cable connection that can slide right on to a motherboard or serial controller header and also provide a through connection for the bottom DB9 ribbon.  Of course fill in any gaps on the PCB for all the throttle blaster components and other internal headers that are needed. 
[image]

If I have the time I might try to prototype something like this. I have limited experience with kicad but I might play around with it this idea.

scrapcomputing, 2024-04-27:
Yeah, that could be a good option for controlling it via the serial port. The current design is more suitable for a 5.25 panel like this: https://www.youtube.com/shorts/n1aFvRNtOcw

skadarnold, 2024-04-27:
That is a great design as well!

skadarnold, 2024-04-27:
I think this can be closed. But wanted to put some ideas out there!

closed 2024-04-27 by skadarnold