scrapcomputing/ThrottleBlaster #11: Remote control / KVM

opualuan, 2024-05-21:
I have several systems going through KVM, I don’t have easy access to the tower to make adjustments. It would be great to have a MHz display and knob at my desk and run a small cable to the PC. Appreciate your ideas on how to extend this without causing noise on the stpclk pin- maybe a second pico in the remote box talking to the one in the pc?

scrapcomputing, 2024-05-21:
Hmm if I were you I would use a UTP cat5 cable (the one used for Ethernet network cables that contains 8 small wires). The display needs 4 cables, one of which is ground, and the rotary encoder needs 5, 2 of which are ground (see the schematic: https://github.com/scrapcomputing/ThrottleBlaster/blob/main/img/ThrottleBlaster.svg). Ground is common across these, so you actually only need 7 wires to carry the signals over. I would use 4 wires for the display and another 4 for the rotary (pins A,C,B and S1).

I hope this helps :)

opualuan, 2024-05-21:
That’s a great breakdown of the signals, cat5 is convenient. I’ll give it a shot, thanks!

scrapcomputing, 2024-05-23:
Another alternative would be to keep the Throttle Blaster on your desk with a long STPCLK cable, but you would also need to power the Throttle Blaster from the PC because both need to have the same ground. I don't think that the STPCLK cable needs to be very short. The frequency of the STPCLK signal is quite low so it should still work with a long cable. Another alternative would be to use a coaxial cable, something like an RG58 should work I guess, this should protect from any noise.

closed 2024-06-08 by scrapcomputing