scrapcomputing/ThrottleBlaster #12: Should modifications to the presets get saved and persist through a power cycle?

Therrixxe, 2024-06-05:
I make a change to a preset and then power cycle the machine and it appears to revert back to the stock value. The CPU max frequency is being saved. I tried it on 3 different machines

scrapcomputing, 2024-06-06:
Thank you for reporting the bug! Yes, the should be saved. I fixed the issue in revision 0.6. Please try out the new firmware version and let me know if you have any more issues. Thanks again :)

Therrixxe, 2024-06-06:
Thank you for the quick fix. Works great now and I also tested out the reverse direction by soldering in a jumper wire ... works perfectly as well. Would it be possible to use the extra GPIO pin on the pico to have buttons that would jump to presets? I was thinking about making a front panel that would have buttons for XT - 4 AT - 8 ,286 10, 386 25, 486 33, Pentium 66. It wouldn't need a new board I could just solder the pin headers to the pico. Thanks again.

scrapcomputing, 2024-06-07:
That's a great idea! Please use the latest firmware (rev 0.7). You can connect the buttons to GPIOs 10,11,12,13,18,19,20,21 and they will select the presets in order (4MHz, 8MHz, etc). GPIO 21 will always switch to the maximum frequency. I have also updated the PCB, just in case :)

closed 2024-06-08 by scrapcomputing