scrapcomputing/ThrottleBlaster #14: Feature request: Reset button using remaining GPIO pins

Therrixxe, 2024-06-07:
Would it be possible to run the reset button from the computer case to the throttleblaster and then from the throttleblaster to the motherboard. When the reset button is pressed it will go to max frequency and then 20 seconds or so later go back to the frequency it was at before the reset button was pressed? 

Currently whenever I press the reset I set it back to stop otherwise it can take a very long time to complete the reboot.

scrapcomputing, 2024-06-08:
That's a great idea. Let me work on it and I'll get back to you.

scrapcomputing, 2024-06-11:
OK I think it's ready. This requires an additional 2N7000 transistor. Its source pin is connected to GP22 (for reset detection) and its gate pin to pin 36 for 3.3V, and its drain pin to the positive pin of Reset (see the schematic). The transistor acts as a level-shifter because the reset pin could be at 5V, while the Pico can only handle 3.3V. When you press the Reset button on your PC you are basically grounding the Reset + pin. This will pull down the transistor's source pin too, resulting in pulling down GP22. The firmware checks the state of GP22 and if it is being pulled down then it will set the frequency to Max for 20 seconds.

Please make sure you connect the reset header correctly (ground to ground and + to +), otherwise your PC will be in constant reset mode. 
I have not done extensive testing, so please report any issues you may run into, and also let me know if you have any questions about the circuit.

Thanks!

Therrixxe, 2024-06-12:
Very cool can't wait to check it out. Soon as the parts show up I will give it a try. You reference RES1 and RES2 however the silk screens on the images you posted don't appear to be labeled. Looks like it doesn't matter which is plugged in as long as you get the polarity right ... it may cause some confusion though.

Therrixxe, 2024-06-12:
I see now that RES1 and RES2 is covered by the pin header in your image.

scrapcomputing, 2024-06-13:
Yeah, I will try to mark them differently in the next revision of the PCB. Thanks for the feedback.

willsy32, 2025-05-13:
I've been testing out the reset button for kicks.  It doesn't always work.  I'll see it do the countdown if I push the button really fast (or deliberately).  Could just be my motherboard.  i mostly run my games in windows 98 with the occasional DOS boot if I need it so it's not a super needed feature for me but figured I'd give some feedback. I'll eventually getting around to serial commands :)

Therrixxe, 2025-05-13:
> I've been testing out the reset button for kicks. It doesn't always work. I'll see it do the countdown if I push the button really fast (or deliberately). Could just be my motherboard. i mostly run [...]

I haven't tried the reset button yet but I have used the serial commands quite a bit using LaunchBox for MS-DOS Special Edition 2. Prior to launching a speed sensitive game ( like Quest for Glory I ) it drops the frequency, launches the game and then executes again when the game is exited to return to full speed.

scrapcomputing, 2025-05-18:
@willsy32 thanks for reporting the issue, I will look into it soon.

kfix, 2025-05-23:
>  It doesn't always work. I'll see it do the countdown if I push the button really fast (or deliberately)

same experience for me (Tualeron socket370 440bx). the mosfet always does its job and relays the press to the motherboard, but the detection on the TB is flaky. I only get the boot countdown if press reset a few times in succession. holding it down for a single long press doesn't trigger the countdown.

willsy32, 2025-05-23:
> same experience for me (Tualeron socket370 440bx). the mosfet always does its job and relays the press to the motherboard, but the detection on the TB is flaky. I only get the boot countdown if [...]

It's the same experience for me and have a socket 1 440bx, your description fits much better and I realized that holding it long isn't what did it but pressing the button a few times

scrapcomputing, 2025-05-25:
This doesn't seem to be a firmware bug. I think it might be a circuit issue. The RESET signal is probably not getting propagated properly through the Q2 transistor. The job of that transistor is to protect the Pico from potentially getting 5V from the motherboard's reset when it's high (which could potentially be at 5V). But I think replacing the Q2 transistor with a diode (like a 1n4001, or any other diode) pin 1 -> pin 3 should do a similar job: the motherboard's 5V can't feed back into the RESET gpio because it's in reverse and when the motherboard's reset is grounded the diode is forward biased pulling down RESET.

[image]

kfix, 2025-05-28:
> But I think replacing the Q2 transistor with a diode (like a 1n4001, or any other diode) pin 1 -> pin 3 should do a similar job

the motherboard still resets (plugged into either header with correct polarity) but now I can't trigger the `boot` countdown at all using a [1N4148](https://www.vishay.com/docs/81857/1n4148.pdf)

[image]

scrapcomputing, 2025-05-29:
Thanks for trying this out. 

Strange... Do you think you could measure the voltage at both sides of the diode with the reset switch in both positions (i.e., pressed and not-pressed) ? 
Thanks!

kfix, 2025-05-29:
with common lead on the anode:
not-pressed: 0.25vdc
pressed: 0.03vdc

scrapcomputing, 2025-05-29:
Thanks for the measurements. 

Could you also try with the pico powered on, the common lead of the multimeter connected to ground and the other lead connected to:
- cathode + reset not pushed
- cathode + reset pushed
- anode + reset not pushed
- anode + reset pushed

What I am trying to figure out is the voltage the motherboard provides on the reset pin on both states of the reset button and how that affects the pico-side too because the pico has a weak pull up, which could be too strong perhaps compared to the motherboard's pull-down when the button is pushed?

kfix, 2025-05-29:
with pico plugged and powered up

| point + condition | VDC |
| ----| ---- |
| cathode + reset not pushed | 5v |
| cathode + reset pushed | 0v |
| anode + reset not pushed | 3.2v |
| anode + reset pushed |  0.45v |

scrapcomputing, 2025-05-29:
Perfect, thanks for the measurements!

So the Pico's pin (connected to the anode) is at CMOS logic high when reset is not pushed and logic low when reset is pushed, just like what the firmware expects.
I would expect this to work hmm... not sure what is wrong. 
If you have an extra Pico could you try replacing it just to rule out damaged GPIO pin? 

Thanks!

scrapcomputing, 2025-05-31:
> That did the trick, thank you!

Great, thanks for trying out the firmware :)

> One issue with the feature is that if you press Reset again while the boot-countdown is in progress, the count doesn't start over.

That's a great point, please try the attached firmware.

[ThrottleBlaster_fixed_reset2.uf2.zip](https://github.com/user-attachments/files/20528334/ThrottleBlaster_fixed_reset2.uf2.zip)

kfix, 2025-05-31:
The new-new firmware works great, _UNLIMITED REBOOTS_!

Thanks @scrapcomputing, feels like I'm trying the new AI vibe coding craze😆