scrapcomputing/ThrottleBlaster #15: Rotary encoder

oXL0C0Xo, 2024-06-12:
I'm wondering if these rotary encoders should work...
https://www.amazon.com/dp/B09KNC1J6H

Because I bought some of them, and when I rotate it, it does not switch the frequency reliably. So additionally I bought another type of encoder:
https://www.ebay.com/itm/164916234390

and this encoder has the same exact issue. What could I be doing wrong?

The encoder does change the frequency, but it seems like it takes a random amount of detents.

I appreciate your time.

Bryon

scrapcomputing, 2024-06-13:
My encoders work kind-of OK, they do skip some positions if rotated too quickly. I think they look like the ones in the amazon link. I have also noticed that using a knob actually helps.
This is a firmware issue that I will try to fix at some point.

scrapcomputing, 2024-11-20:
Firmware rev 0.8b should address the rotary encoder issue. Please give it a try when you get the chance.