scrapcomputing/ThrottleBlaster #17: No effect when changing frequency

dbenoy, 2024-09-26:
My hardware: MSI MS5182 Socket 7 Motherboard, AMD K6/2-475

I'm getting no effect when I change the frequency. I'm new to electronics so bear with me :)

To verify the Throttle Blaster was working, I pulled up the STPCLK# output pin by connecting it with a 10k resistor to the display's VCC and examined it with an oscilloscope. It appeared to be pulling the signal low.

Also, when I connect the STPCLK# pin on the CPU to ground directly, the CPU pauses as expected.

I also set the clock speed all the way down to 0, and then measured between the STPCLK# output pin and ground and got roughly 100 ohms. I see that this should be expected because of the 100 ohm resistor. Oops. Should I not have been testing by connecting the CPU pin directly to ground like I did? Is that 100 ohm resistor required to prevent damage?

scrapcomputing, 2024-09-26:
The STPCLK# pin gets pulled down to the Throttle Blaster's ground by the transistor. Since you already checked that STPCLK# does work (with the help of the 10K pull-up resistor), the only thing that I can think of is that the Throttle Blaster's ground may not be the same as the PC's ground. This can happen if you are powering the Pico via USB using an external power adapter. If you use the Throttle Blaster's 4-pin power connector that shouldn't be an issue. In any case I would try connecting your PC's ground with the Throttle Blaster's ground using a piece of wire to see if this makes any difference.

dbenoy, 2024-09-26:
Yep I was powering it through the PC and got continuity between the Throttle Blaster ground and the PC ground.

I just tried testing it by connecting the CPU pin to ground through a 100ohm resistor, bypassing the Throttle Blaster entirely, and it didn't work. :sweat_smile: Direct connection still pauses the CPU, but going through a 100 ohm resistor doesn't. I measured the voltage between the CPU pin and ground and it drops to 1.7 volts which isn't low enough, yeah. Color me confused.

dbenoy, 2024-09-26:
(I may be out to lunch. New to electronics) My theory is that the internal pullup resistance on this CPU is somewhere in the neighborhood of 100 ohms, and so a 100 ohm resistor on the pin would divide the voltage by half and give the results I'm seeing. And I should try a lower resistance.

But I'm scared to do that without hearing what you think first :p

scrapcomputing, 2024-09-26:
This is very strange, 100 Ohms should be more than enough to pull down the STPCLK# pin. I am wondering whether it is the CPU itself that pulls it up hard or the motherboard. Could you please measure the resistance between the STPCLK# pin and VCC3 (or is it VCC2, can't remember) on the CPU with the CPU removed. I think the strong pull up should most probably be on the motherboard side, so it's probably also worth measuring the resistance on the socket with the CPU removed, and perhaps the voltage when it gets powered on?

Not sure it's safe to reduce the resistance, but if you really want to try it, add another 100 Ohm resistor in parallel, but it's probably best to try to understand what is causing the issue first.

dbenoy, 2024-09-26:
I'm getting like 10 megaohms between STPCLK# and the VCC3 and VCC2 pins. I tried switching the probes and it was some 5 megaohms. So basically no continuity?

dbenoy, 2024-09-26:
The voltage between STPCLK# and ground when the system is powered on is 3.27V, and I measured before that it only drops to 1.7V when it goes through a 100 ohm resistor.

When I short it to ground (and it pauses the CPU) through my multimeter, the current is less than a milliamp... which is not what I'd expect if the internal pullup was 100ohms .. hm. 3.3v / 100ohm = 33mA

dbenoy, 2024-09-26:
I wish the datasheet were more clear on exactly how much resistance to ground is too much according to its design. But I can't find anything to that effect.

dbenoy, 2024-09-26:
I pulled the CPU out as suggested. If it's connected to nothing at all on the motherboard then I imagine that would be a completely open connection.

But it isn't. I still get the 5Mohm reading between STPCLK# and ground and I can see like 0.5V in diode mode on my multimeter (I'm still not sure what I'm doing but I think that's indicating some kind of logic is connected there?)

I hit up the datasheets for the southbridge and it turns out it is connected to the STPCLK# pin:
[image]

... I guess that answers what's going on. The motherboard probably is pulling that pin high. Sad trombone for me.

closed 2024-09-26 by dbenoy

scrapcomputing, 2024-09-27:
The chip is expected to control the STPCLK pin, but they would usually only pull it.

I think there are two ways to get it to work:
1. You can just try lowering the 100 Ohm resistor until it works. There is a chance that this will damage the chip but I would guess that there should be some protection against short circuits, so it might work just fine.
2. You can try cutting the trace that connects the motherboard's chip to the STPCLK# pin at the socket.