scrapcomputing/ThrottleBlaster #19: Future changes question for potential redesign

skadarnold, 2024-11-19:
Hey Scrap! 

Now that I have been getting some experience with PCB design I have been venturing into different projects and planning out designs I have thought of over the years. I have had some ideas for changing the layout of the ThrottleBlaster for making a PCI bracket mounting revision. My 5.25" drive bays are full right now, and instead of switching them out I thought this would be a good option to have instead. I also have some ideas for bringing the display and control panel to an external module to sit on the desk connected via 9-pin d-sub connector or something similar. Not sure if I will do the control panel since I plan to mostly use the UART for control anyway, but it depends on how ambitious I get. 

My main question is to ask if you have any future revision plans that I should be prepared to adjust to if I start making changes on my own fork? I don't mind adapting to small changes you might have in the PCB or schematic, but once I start modifying the layout of everything I am not sure how easy it will be to merge any new changes without dealing with conflicts. I feel like things are pretty settled on your end for this project, but I just thought I would check before diving in to deep. 

Also, since I am also close to using up my PCI / ISA brackets at the moment I have also been considering combining the ThrottleBlaster into some other project ideas I have. Since the ThrottleBlaster takes up very little space, I might work to have it as a feature in an all in one card.

scrapcomputing, 2024-11-20:
Hi @skadarnold,
I don't have any plans for any major changes.
A pci version would be really nice, I remember some folks asking for such a variant.

skadarnold, 2024-11-26:
Here is a little sneak peak of what I am brewing up. Originally I thought it was going to be mounted to a PCI bracket, but I decided to move it to a full on ISA slot so it could include some other features as they develop. 

[image]

I still have a ways to go with this before I test it out, but I think it is coming along. I will hold off on going over features other than the throttle blaster for now since they are still getting ironed out. But you can see the ThrottleBlaster section to the left. The idea the core of the ThrottleBlaster will be controlled via RS232, with the option to use an addon that connects with a DB9 type header. I also added two DB9 headers to attach to the MB serial headers, one to be used for input and the other for throughput so that it could still allow for a DSUB connector. The controller addon circuit is the same layout and size as the original ThrottleBlaster so that it could still mount into currently designed mounting, example like the 5.25" bay mounting kit. It also allows for the option to use the STPCLK pin in case it is to be mounted in a better position to the CPU. Also, I added a horizontal mounting EC11 rotary encoder so that it could easily be attached to a PCI bracket with a 1/4" hole. This is all optional since my main purpose is to control everything via serial, but it would be nice to be able to have tactile buttons and dials for this. Since the remote addon is connected via DB9 style header, I also plan to create a desktop control unit that can extend the DB9 connection with a standard serial cable.

scrapcomputing, 2024-12-12:
It looks really cool!

RetroLoom, 2025-01-31:
I haven't forgotten about this project but I am taking a few side steps in developing the main board. I have created this PWM to STPCLK module inspired by the Throttle Blaster. 

[PWM2STPCLK Project](https://github.com/RetroLoom/PWM2STPCLK)

The idea is to make the throttling circuit compatible with any PWM controller and to have a secured position mounted in the computer. I plan on revising the Throttle Blaster to be compatible with this configuration as well. It can be set up using a PWM fan controller and connected with a  PWM fan cable as well.  

I thought you might be interested in something like this as well.

scrapcomputing, 2025-02-06:
Nice! This looks awesome! I didn't know you could actually buy an off-the-shelf PWM module!

Sure please keep me updated with your latest design :)

RetroLoom, 2025-02-06:
> Nice! This looks awesome! I didn't know you could actually buy an off-the-shelf PWM module!
> 
> Sure please keep me updated with your latest design :)

Thanks! This is the PWM fan controller I used in the demonstration photos. 

[PWM Fan Controller](https://www.aliexpress.us/item/3256805787442361.html?spm=a2g0o.order_list.order_list_main.17.719d1802suoon6&gatewayAdapt=glo2usa)

I think you can find similar ones from amazon as well, but the Aliexpress ones are very inexpensive. I will update this on the readme as well. 

I will keep you in touch!

RetroLoom, 2025-02-17:
I created an [issue](https://github.com/RetroLoom/PWM2STPCLK/issues/1) on my PWM2STPCLK repo that I wanted to mention to you here. I have noticed similar switching noise with the ThrottleBlaster and wouldn't mind exploring the reasons and possible solutions for it. 

For isolating the grounds, I thought about replacing the 2N7002 with a [TLP175A photo-MOSFET](https://www.mouser.com/ProductDetail/Toshiba/TLP175ATPLE?qs=HVbQlW5zcXXp9ILK7hvmMw%3D%3D), but I have not worked with one before so I thought it would be good to have more minds on it. 

Here is a sketch of a schematic I was thinking
[image]

Let me know what you think and thank you.

scrapcomputing, 2025-02-19:
Hmm could it be that the switching noise is from the long wire ? Perhaps if you replaced it with a coaxial cable instead it might reduce it ? I haven't noticed any such noise, though I haven't tried many sound cards.

RetroLoom, 2025-02-19:
I don't think it is coming from the long wire, but seems to originate from the PSU itself. I think it can be dependent on the quality of the ground traces / plane or even the the quality of the PSU so I don't know if it is a problem with all systems / configurations. I will try and do a little more testing to see if I can confirm. But so far in my research I have learned this can be common with fast switching. For an example even the pull-down from fast switching the voltage on the STPCLK pin can cause noise on certain PSU's or maybe even in the CPU itself. Or it could even be creating EMI that radiates from the CPU area. I am no expert and this is all mostly speculation but I thought I might explore the concept some more. If any of these conditions are true it seems the best solution is to ground the STPCLK pin directly to the chassis. 

I will do some more testing and let you know what I discover.

RetroLoom, 2025-02-23:
I am still doing some research on this and plan to try in a different system to see if I have similar results. This system doesn't have a particularly nice PSU so it could be related to poor quality. When I get my  ear up to the noise it is definitely originating from the PSU. I decided to post a video so you can hear what I am talking about. 

https://github.com/user-attachments/assets/737234e5-fb8f-4025-a86b-178b7385e801
You can hear when I click the PWM controller on the noise activates, then mellows out once I am at full on. It sort of reminds me of noisy dimmer light switches where the noise generates in between full dim to full brightness. This is using the PWM2STPCLK but I had similar noises when using the ThrottleBlaster. I had hoped mounting directly to the chassis might help mitigate some of this alone but apparently it didn't change much. 

I don't know if it is radiating EMI from the CPU and it is being picked up by the PSU, or if it is back feeding through ground and oscillating circuits on the way, or something else. The radiating might be the most concerning if that is the case though.

RetroLoom, 2025-02-23:
I just tested this with my main system and it does not generate any noise. It does have a better quality PSU. While this is good news overall I still wonder if it is still worth exploring exactly what causes this oscillation in some configurations.

RetroLoom, 2025-04-13:
This thread can be closed. I will open the noise issue with a solution in a new thread.

closed 2025-04-13 by RetroLoom