scrapcomputing/ThrottleBlaster #20: On motherboard with Socket 775 does not work

nahimov, 2025-01-09:
Hello! Could you help me understand why it doesn't work on socket 775? I use #STPCLK on Socket 775 pin "M3" but have no luck.

nahimov, 2025-01-09:
I think that we find how to solve this issue, we need some days for test.

scrapcomputing, 2025-01-10:
I have not tried it on a socket 775 system. I am interested to know whether it works.

nahimov, 2025-01-13:
Hello! It was launched on Anovo AIMB-865 (I865+ICH5). A 62ohm resistor (instead of 100 ohms) was used for operation. The system is working, but in speedsys, not one of the prepared profiles does not fit into the schedule. Too fast. Programs do not load with a pwm period below 17-19 (may it affect the operation of Pio mode?)
The encoder is unstable. The problem is with the profile mode, when setting the frequency or PWM, the control falls off. It does not respond to push or encoder turns in any way. The encoder itself is obviously worki

scrapcomputing, 2025-01-13:
- The 100 Ohm resistor at the STPCLK pin is just used for safety, but any lower value should also work.
- You can modify each preset to suit your needs. Just long push on it to adjust it. Regarding the pwm issue, it  may be related to the STPCLK resistor. Could you try a lower resistor value than 62 ohms, or perhaps bridge the resistor with a piece of wire so it's effectively 0 Ohms?
- This sounds like a firmware bug. Could you provide me with reproduction steps so I can fix it ? Thanks!

nahimov, 2025-01-13:
The encoder works fine until you go into editing presets.
1. Select any of the ready-made profiles and press the long push on the encoder.
2. Select the desired MHZ (Pre) and press the push button to confirm.
3. Select the desired PWM (Per) and press the push button to confirm.

After selecting PWM (Per) and pressing the encoder button, it may freeze in several ways:
* Freezes at the last selected PWM value, there is no reaction from the encoder. The numbers on the screen
keep blinking. (you have to reboot ThrottleBlaster) 
* Freezes after the done message appears, and again there is no reaction from the encoder.
(you have to restart ThrottleBlaster)

We need some days for testing...

scrapcomputing, 2025-01-20:
Thanks for the detailed bug report. I fixed the firmware bug. Please upgrade the firmware to the latest release: Rev 0.8c.

nahimov, 2025-01-29:
Hi! First of all, thank you for latest firmware.

About problem on socket 775

During testing, there were big problems with the disk subsystem (IDE and SATA). Any decrease in frequency immediately fails. Pin M3 is responsible for the operation of STPCLK. Tracking where it goes to the south bridge and isolating it from the south bridge did not give a positive result, apparently the signals to the south bridge also go through other pins. Everything was tested on ICH5 (Intel 82801EB). When using an external controller, everything is working normally. I think this problem will also be on socket 478. We checked on two different boards, all of which had this south bridge.

The verdict:

1. The operation of the Trottblaster negatively affects the operation of the integrated disk subsystem.
2. The operation of the Trottblaster will be stable if you use an external disk controller.

scrapcomputing, 2025-02-06:
Thanks for the update. 

I am curious, how did you attach a wire to the STPCLK pin? Isn't socket 775 surface mounted?

nahimov, 2025-02-07:
Hi! First of all... Nope :-)

[image]

[image]

And second, we finded contact who go to southbridge on our motherboard

[image]

nahimov, 2025-02-07:
Anyway you can find this pin by use multitester and find near south bridge

[image]

Also we do it on another motherboard 775i65G

[image]

scrapcomputing, 2025-04-07:
All you need to do is change the entries of `const std::vector<Entry> ImmutableTable` in `src/Presets.h`  and rebuild.

I built one (attached), though I don't have time to test it right now. But I think it should work. Let me know if it doesn't.

[ThrottleBlaster_4.77_12_40_133_max.uf2.zip](https://github.com/user-attachments/files/19634351/ThrottleBlaster_4.77_12_40_133_max.uf2.zip)

nahimov, 2025-04-08:
Thank you very much! I will try.

filthyfreak, 2025-05-01:
Hi @nahimov 

I plan on setting up TB on my socket 478 which has the same Intel 82801EB (ICH5).

Which external storage controller are you using and how are you booting into it?

nahimov, 2025-05-29:
Hi @filthyfreak 
Sorry for delay. It doesn't matter which controller. I used the regular promise ata 100 ide. In Bios you have menu for boot from controller.

nahimov, 2025-05-29:
By the way, if anyone is interested, I did this on Tualatin-s.

[image]

[image]

[image]

nahimov, 2025-06-21:
@Peckmore Hi! Right now I can't. I try at next month. This files on another pc. But you have to understand that I made it for a specific screen and specific buttons I bought.
p.s. It's very strange, but no one had noticed this panel before. Just like no one pays attention to my implementation of the panel for mt32-pi, which is higher in the photo. I tried to choose a screen that looked as much like the original mt-32 as possible....

nahimov, 2025-06-21:
@scrapcomputing Hello, my friend! I found one problem. I bought a second pi for a different throttle and what can I say. The 0.8c version of the software works well on the left pi in the photo (dark green), but not on the right pi (fake Chinese, as I understand it) because I can't save the frequency, but the older 0.8b version works well on this device. :-) Did I understand correctly that this can be fixed in version 0.8d, and both pi will be able to work properly? 

[image]

scrapcomputing, 2025-06-21:
Hi @nahimov
- 0.8d fixes an issue related to reset detection. So if you are not using that feature then this release shouldn't fix your issue, but you could try it. Please let me know if this doesn't work on the clone.
- 0.8c fixes a crash that would hang the ThrottleBlaster when you would try to save the frequency (introduced in 0.8b). Hmm I am not sure why this wouldn't work on a clone.
- 0.8b contains a bug that crashes the TB which is why I removed it from the repo, but if you think that one worked fine here it is (attached)

[ThrottleBlaster_0.8b.uf2.zip](https://github.com/user-attachments/files/20847542/ThrottleBlaster_0.8b.uf2.zip)

nahimov, 2025-06-22:
@scrapcomputing Yes, I'll post them when I can, both of them (for mt32 and for trottle on another PC). But you need to understand that they were made for certain screens and buttons that I bought on Aliexpress. Do I need to post links to screens and buttons? 0.8b on the clone set the frequency. 0.8c on the original set frenquency, I don't know why that is 🤷🏻

scrapcomputing, 2025-06-22:
@nahimov Using your specific buttons is perfectly fine. Just please make sure you share the sources of the design (not just the STL file) in case others want to make small modifications to suit their buttons or to accommodate future ThrottleBlaster PCBs that may not fit the current one.

I don't think links for screens and buttons are necessary (though that would be nice), but some close-up photos and perhaps some search keywords would be super helpful.

nahimov, 2025-06-30:
@Peckmore @scrapcomputing Hello!

[throttle.zip](https://github.com/user-attachments/files/20976616/throttle.zip)

[mt32-pi.zip](https://github.com/user-attachments/files/20976619/mt32-pi.zip)

I don't know you can or not buy from Aliexpress at your country.

screen for throttle - https://aliexpress.ru/item/1005008137384657.html

[image]

button for throttle - https://aliexpress.ru/item/1005007560430185.html

[image]

screen for mt-32pi - https://www.aliexpress.ru/item/1005006632495660.html

[image]

driver for screen - https://www.aliexpress.ru/item/1005006086057327.html

[image]

button for mt-32pi - https://aliexpress.ru/item/1005006689494738.html

[image]

One moment. Because screen is big I used separate power supply cable. So SDA and SCL I used from mt-32pi but power external (As an option from the floppy drive, you can use). On button I used all from mt32-pi and power also.

nahimov, 2025-06-30:
A lot of old motherboards from the time of pentium 2 and pentium 3 have an IrDA connector, the TTL level is used there, I tried to make the trottle control work through it, but unfortunately nothing worked. As it is, it would be great not to solder to the com port.

scrapcomputing, 2025-06-30:
@nahimov thanks for sharing photos and links :)

As for your IrDA question, I am not familiar with the IrDA connector protocol but I can look into it.
- What software are you using from the PC side to talk to it ? 
- Also since it's TTL it would require level shifting on the Pico side in place of the MAX3232 (which is not TTL). How did you connect the IrDA port to the Throttle Blaster?

nahimov, 2025-06-30:
@scrapcomputing I'll try to use kermit. I solder pins to picopi not to max232 ofcourse. On motherboards wehave pins for IrDa and in bios we can set use IrDa and IrDa are set address from com1 or com2 (3F8 or 2F8). So as I can understand after it we can use IrDa like UART and if we solder to pins on picopi it must start work, but have no luck, I havent Oscillograph, maybe we can use some kind of resistors to match the signal.

p.s. did you see two archives with panels ?

scrapcomputing, 2025-07-01:
I don't think kermit will work. It expects a UART, not sure the interface of the IrDa chip but if it's not UART it is unlikely to work.

Hmm why can't you just plug in to the COM pins on the motherboard? Most motherboards should have those, no need for soldering.

> p.s. did you see two archives with panels ?

I haven't checked them out yet. But you seem to have done some great work, why don't you create a github page to show it off ?

nahimov, 2025-07-01:
About github, I don't have anything else interesting, so I probably won't :-) 

What about IrDa. That's what Artificial Intelligence told me: "Yes, IrDA (Infrared Data Association) SIR (Serial Infrared) mode relies on a UART (Universal Asynchronous Receiver/Transmitter) for its physical layer. IrDA SIR uses the same data framing as RS-232, which is typically handled by a UART. The Uart serial data stream is then encoded and decoded by an IrDA encoder/decoder (ENDEC) module before being transmitted via infrared light."

I tried connecting from my laptop via usb - uart on PuTTY and listen, But I didn't catch anything.

About plug in to the COM pins. We can do this only on older board like AT, on ATX it's hard to use, because COM ports on back plate pannel. On my PC, which I showed above, I soldered to the connector.

[image]

Unfortunately, my knowledge is not enough to understand and deal with this. Of course, I understand that if you don't have time for this, it's not a big problem, but this connection may be useful for some other projects in the future.

p.s. By the way, my friend, what do you think about the zuluide project? This is the first CD rom ide emulation project with the Possibility to use the red book, i.e. bin/cue that play through the audio output. It's a very interesting thing, the software is publicly available, but the hardware part is expensive, it seems to me that it can be implemented in a cheaper version on a picopi and not with a SD card, but with a converter to CF or SSD on SATA if you have very much images of cd.

scrapcomputing, 2025-10-04:
@nahimov Thanks for sharing your success with the Pentium pro! I will certainly update the page, but could I ask you to share 2 photos, one from the socket side and one from the back. Ideally I would like them to be taken from a similar angle as the photos of the other sockets listed in the main page.

Regarding talking to IrDA, this looks very interesting. If there is a library for the pico that implements the protocol we could easily plug it in.

nahimov, 2025-10-04:
@scrapcomputing good?

[image]

[image]

scrapcomputing, 2025-10-04:
@nahimov these look great, thanks! I will crop them a bit and will include them. Oh by the way, what license are you sharing them under ? or should we treat them as "public domain" ?

Regarding the usb-c pico clone, I was not aware that these were not powering the throttle blaster! The actual throttling should still work as it's powered by the pico's 3.3v output. The 7-segment display is getting powered by the VSYS pin, so perhaps they have not connected that? Strange.

nahimov, 2025-10-04:
[image]

[image]

nahimov, 2025-10-05:
@scrapcomputing I found another one written in C, not Python. Just for an appointment. In general, this is exactly what we are interested in, because there is nothing to transmit from the trottleblaster, we need to send it to it. https://github.com/kazkojima/pico-irda

nahimov, 2025-10-08:
@scrapcomputing Hello! Have you checked the links about irda, they won't work?