scrapcomputing/ThrottleBlaster #22: Slot 1 Pentium3 800mhz not working

willsy32, 2025-05-09:
Soldered to a via that connects to B6 on my pentium 3.  Nothing seems to work and I think it's because the trace doesn't go straight to the cpu but instead through a voltage translator "GTL2000DL" (https://www.nxp.com/part/GTL2000DL).  I assume I'll need to connect directly to the back of the cpu, but can't find a schematic (hard to follow the trace).  I can provide more pics if needed (back of cpu is provided), and would love any help I can get. Thanks!

[image]

scrapcomputing, 2025-05-09:
I don't think the level shifter would be an issue. It's job is to translate 3.3V motherboard-based signals down to ~1.5V for the CPU and vice versa. The throttle blaster would still pull down the 3.3V STPCLK# signal to zero on the motherboard side and that should still generate the corresponding pulses from ~1.5V down to zero on the CPU side.

The first thing I would check is whether there is continuity between the ground of the Throttle Blaster and that of the CPU. This means that there should be a wire connecting the ground of the Throttle Blaster and the ground of the motherboard or the PC case chassis. Without it the circuit won't work.

If this doesn't solve the problem, you couldmeasure the voltage at B6 with a multimeter. It should normally be high (3.3V or higher) when the Throttle Blaster is disconnected or when set at the maximum "frequency" setting. Dialing it down should decrease the average voltage level, bringing it almost down to 0 at the lowest setting.

Finally you could try soldering the wire to the low-voltage section of the level-shifter. It should be the pin on the opposite side if I am reading the datasheet for the GTL2000 correctly. But I doubt that would solve the issue.

I hope this helps, let me know if you are still having issues with it.

Thanks!

willsy32, 2025-05-09:
I'm grabbing ground from my psu and just have my floppy power connector connected directly to my throttle blaster.  Ill check the voltage to see if it's being pulled down, I appreciate the help!

willsy32, 2025-05-10:
Ok, so I reflowed all of the solder joints and then tested the voltages.  At 800mhz it's at 2.5 volts and at its lowest it's at approximately 0.8 volts.  Haven't done any benchmarks or anything yet since I took out my PC to the workbench, but I'm happy it's doing what it's doing something.  For some reason though at 25mhz it's at 0.84 volts so not sure if it's scaling right or not

willsy32, 2025-05-10:
I haven't tested any bench marks because I don't have a monitor where I'm at, but I want to confirm if those values are good or bad before I move forward

scrapcomputing, 2025-05-10:
I think the voltages sound reasonable.

The easiest way to test it is with the DOS benchmark pack from philscomputerlab.

willsy32, 2025-05-10:
Great, I'll test in the morning!

willsy32, 2025-05-10:
Yeah, the bios isn't posting....

willsy32, 2025-05-10:
After testing a few benchmarks and confirming the voltage being pulled to 0.8. I am not seeing any slowdowns in the every benchmark test from philscomplab dosbenchmark pack

willsy32, 2025-05-10:
I am not seeing any slowdowns (even minor) when pulling 2.5volts to 0.8volts.  I've tested all of the philscomplab benchmarks and tested 800mhz and 0mhz

scrapcomputing, 2025-05-10:
Ah great you got it working!

Strange, the slowdown should be more or less linear. One thing you could try is to reduce the value of the R2 resistor at the output transistor from 100 Ohms to perhaps half (e.g. by soldering another 100Ohm resistor in parallel). The resistor is there mainly for safety, to avoid pulling down the STPCLK pin too hard. But an average value of 0.8V at 4MHz is probably a bit higher than expected (even though measuring with a multimeter is not accurate), so perhaps the signal is not pulled down as hard as it should.

I hope this helps.

willsy32, 2025-05-11:
This may have done it.  I got this throttle blaster pack from an eBay seller and didn't double check the values on the actual resistor.  Seems R2 has been 1000 ohms all along.  I'll 

[image]

willsy32, 2025-05-11:
Lol not easy to do. Almost gave up to try 47 ohm instead XD

[image]

willsy32, 2025-05-11:
And it works!!!!!!

willsy32, 2025-05-11:
Works absolutely amazing too.  Super happy to finally get this project done with :)

closed 2025-05-11 by willsy32

scrapcomputing, 2025-05-11:
That's great! Happy throttling :)

As for the accuracy you can fine-tune each preset by changing the actual frequency (not the displayed one) and its PWM period (see the state diagrams at the project's page).

47 Ohms should work too. You could also stack the 100 Ohm resistors on top of each other, it might be easier to solder that way.
Btw I will add a note about the resistor value at the project's page. Thanks for helping with troubleshooting :)